RTimothyEdwards/magic #322: -O2 compilation causes the Match function in utils/Match.c to match partial string.

d-m-bailey, 2024-08-27:
This issue is just for informational purposes.

Compiling with `-O2` optimization causes the pattern string to match any string beginning with the pattern string.
`astring` will match `astring_and_anything_else`.

Don't use `-O2`.

dlmiles, 2024-10-06:
Unable to reproduce.

Please advise the platform, distribution (/etc/os-release) and compiler (gcc -v) details.

dlmiles, 2024-10-06:
[match_main.c.txt](https://github.com/user-attachments/files/17271115/match_main.c.txt)

checkout, then configure magic project
then use command line at of file to compile testcase

please edit test case to demonstrate the claim in this bug and report platform details

d-m-bailey, 2024-10-10:
@dlmiles Well, I can't get it to work (by work I mean fail) with the snippet of code you provided. I'll see if I can recreate the problem with the original program.