RTimothyEdwards/magic #324: showing parameters window using bind keys isn't working

areda0, 2024-09-03:
[image]

cannot get the parameters window of a selected device to show-up using the bind key "ctrl+p" . attached is the error message on magic console.

d-m-bailey, 2024-09-03:
@areda0 What command and options did you use to start magic?

areda0, 2024-09-04:
@d-m-bailey  
magic -d XR -rcfile $PDK_ROOT/$PDK/libs.tech/magic/sky130A.magicrc

d-m-bailey, 2024-09-04:
@areda0 That should work as long as PDK is `sky130A`. If you use
```
magic -d XR -rcfile $PDK_ROOT/$PDK/libs.tech/magic/$PDK.magicrc
```
it should work with `sky130B` also.

areda0, 2024-09-04:
the PDK is correctly defined.. using sky130A and still cannot get the bind key to work as expected. 
side note: I installed magic from source

RTimothyEdwards, 2024-09-26:
I seem to have managed to mess up the "cellname property <cell_name> <key>" command, which is no longer returning a value for the property.

RTimothyEdwards, 2024-09-26:
Presumably this one.  I made corrections to the "property" command syntax parsing, but did not consider that the same code is called from the "cellname property" command, with a different index into the command arguments.  I will get this fixed ASAP.  Sorry for the delay.

```
commit b441d668bd11e261a61c999c6c5ab55ffcc0da5f
Author: Tim Edwards <tim@opencircuitdesign.com>
Date:   Sat Aug 3 12:05:41 2024 -0400

    Corrected the "property" command parsing, which had a print statement
    without an argument, resulting in random things getting printed to
    the output when a property name does not exist.
```

RTimothyEdwards, 2024-09-26:
Fixed in magic version 8.3.492 (see commit notes).
@areda0 :  Thank you for logging the issue.

closed 2024-10-02 by RTimothyEdwards