RTimothyEdwards/magic #339: DBtpaint2.c:529 please review of 'break' keyword location

dlmiles, 2024-10-07:
https://github.com/RTimothyEdwards/magic/blob/master/database/DBtpaint2.c#L529

Please review the indentation of this  `break` keyword, as it makes the containing `for()` loop redundant and reduces it to an `if(TT_TECHDEPBASE < DBNumUserLayers)` statement.

Maybe a comment left in the area to say this is ok/intentional with some insight might help the next reader.

SonarCloud
Change this loop body so that it can be executed more than once.
https://sonarcloud.io/project/issues?open=AZJB16rNNGfDNup0RiZp&id=dlmiles_magic

RTimothyEdwards, 2024-12-26:
Clearly wrong, thanks.  I will fix it.  I'm not even sure the `break` should be there at all because it's possible to define a "stacked" contact type that always exists on more than two planes.

RTimothyEdwards, 2024-12-26:
Fixed in magic version 8.3.509.  I'm not aware of any problems having to do with this code, so I have no idea what impact it has, if any.

closed 2024-12-26 by RTimothyEdwards