RTimothyEdwards/magic #353: SIGSEGV extresist ResNewSDDevice()

dlmiles, 2024-11-10:
Please check the script https://github.com/algofoogle/tt09-ring-osc/commit/9971449499949674c0413245df31997f05faafdc#diff-521581d2cd10f21d7e928d1436d258208e077a050709787a451bb3c59cab5c75  for any obvious errors in usage

Project and version https://github.com/algofoogle/tt09-ring-osc/commit/9971449499949674c0413245df31997f05faafdc
cd ext
gdb --args /usr/local/lib/magic/tcl/magicdnull -noconsole -dnull tcl/extract_for_sim.tcl tt_um_algofoogle_tt09_ring_osc

```
#0  0x00007fffe7eb2d45 in ResNewSDDevice (tile=0x7ffff7f595f8, tp=0x7ffff7f59da0, xj=27403, yj=21634, direction=1, PendingList=0x7fffe8231550 <ResNodeQueue>) at ResJunct.c:59
#1  0x00007fffe7ebf6f4 in ResEachTile (tile=0x7ffff7f595f8, startpoint=0x7fffffffbc68) at ResBasic.c:329
#2  0x00007fffe7eaf1b1 in ResProcessTiles (goodies=0x7fffe8c486a0 <gparams>, origin=0x7fffffffbc68) at ResMain.c:510
#3  0x00007fffe7eb1005 in ResExtractNet (node=0xad5450, goodies=0x7fffe8c486a0 <gparams>, cellname=0x169aa50 "tt_um_algofoogle_tt09_ring_osc_parax") at ResMain.c:1203
#4  0x00007fffe7ebd4af in ResCheckSimNodes (celldef=0x11fec10, resisdata=0x7fffffffbf80) at ResRex.c:1207
#5  0x00007fffe7ebae40 in ExtResisForDef (celldef=0x11fec10, resisdata=0x7fffffffbf80) at ResRex.c:187
#6  0x00007fffe7ebc5a4 in CmdExtResis (win=0x7a0670, cmd=0xc55e70) at ResRex.c:684
#7  0x00007fffe7ea7d86 in WindExecute (w=0x7a0670, rc=0x7abf90, cmd=0xc55e70) at windMain.c:418
#8  0x00007fffe7e18385 in DBWcommands (w=0x7a0670, cmd=0xc55e70) at DBWprocs.c:689
#9  0x00007fffe7ea4902 in WindSendCommand (w=0x7a0670, cmd=0xc55e70, quiet=1 '\001') at windSend.c:263
#10 0x00007fffe7e9cac3 in TxTclDispatch (clientData=0x0, argc=1, argv=0x621a10, quiet=1 '\001') at txCommands.c:1391
#11 0x00007fffe7fc388e in _tcl_dispatch (clientData=0x0, interp=0x60e960, argc=1, argv=0x621a10) at tclmagic.c:43
```

```
(gdb) frame 10
#10 0x00007fffe7e9cac3 in TxTclDispatch (clientData=0x0, argc=1, argv=0x621a10, quiet=1 '\001') at txCommands.c:1391
warning: Source file is more recent than executable.
1391        if (DRCBackGround != DRC_SET_OFF) DRCBackGround = DRC_NOT_SET;
(gdb) p argv[0]
$1 = 0xc84410 "extresist"
(gdb) p argv[1]
$2 = 0x0
```

The issue is deref of an invalid memory location `0xc000000000000004` as in CLIENTDEFAULT or MINFINITY:

```
$25 = (Tile *) 0x7ffff7f59da0
(gdb) p *tp
$26 = {
  ti_body = 0x18,
  ti_lb = 0x7ffff7f5a078,
  ti_bl = 0x7ffff7f595f8,
  ti_tr = 0x7ffff7f5a3f8,
  ti_rt = 0x7ffff7f5a200,
  ti_ll = {
    p_x = 27403,
    p_y = 21555
  },
  ti_client = 0xc000000000000004
}
(gdb) p devptr
$27 = (ExtDevice *) 0x1f61740
(gdb) p *devptr
$28 = {
  exts_deviceName = 0x1ed32b0 "sky130_fd_pr__special_pfet_01v8_hvt",
  exts_deviceParams = 0x8fd370,
  exts_deviceClass = 13 '\r',
  exts_deviceResist = {
    ht_table = 0x850110,
    ht_size = 8,
    ht_nEntries = 0,
    ht_downShift = 27,
    ht_mask = 7,
    ht_ptrKeys = 0,
    ht_copyFn = 0x0,
    ht_compareFn = 0x0,
    ht_hashFn = 0x0,
    ht_killFn = 0x0
  },
  exts_linearResist = 9,
  exts_deviceSDTypes = 0x94df70,
[... 143 lines cut ...]
60	    if ((j->sourceEdge & direction) != 0)
61	    {
62		if (resDev->rd_fet_source == (resNode *) NULL)
63		{
(gdb) p tp
$56 = (Tile *) 0x7ffff7f59da0
(gdb) p *tp
$57 = {
  ti_body = 0x18,
  ti_lb = 0x7ffff7f5a078,
  ti_bl = 0x7ffff7f595f8,
  ti_tr = 0x7ffff7f5a3f8,
  ti_rt = 0x7ffff7f5a200,
  ti_ll = {
    p_x = 27403,
    p_y = 21555
  },
  ti_client = 0xc000000000000004
}
(gdb) p j
$58 = (tileJunk *) 0xc000000000000004
(gdb) frame
#0  0x00007fffe7eb2d45 in ResNewSDDevice (tile=0x7ffff7f595f8, tp=0x7ffff7f59da0, xj=27403, yj=21634, direction=1, PendingList=0x7fffe8231550 <ResNodeQueue>) at ResJunct.c:59
59	    resDev = j->deviceList;
```

dlmiles, 2024-11-10:
It is the value of `j` (not `j->deviceList`) that is the issue.
The value is `0xc000000000000004` which comes from `TiSetClient(newtile, CLIENTDEFAULT);`  this is the value of MINFINITY or related data value `#define	INFINITY	((1 << (8*sizeof (int) - 2)) - 4)`  which is a value of `bffffffffffffffc`

Which works out like `perl -e 'printf "%x\n", -((1 << (8*8 - 2)) - 4);'` for a result of `c000000000000004`

The traces show the specific memory location having its `ti_client` changed a few times during the processes of the TCL script.  With the last time before the crash being from `DBResetTilePlane()`

RTimothyEdwards, 2024-11-10:
Sorry, I don't have enough time to look at this in enough detail to make decently informed statements about it.  But the basic underlying issue appears to be that `ExtResetTiles()` was called from `ResExtractNet()` to remove all the "`tileJunk`" client data records from the tiles in the database, but then `ResExtractNet()` continued to process tiles.  Checking for CLIENTDEFAULT would be a quick way to stop the segfaulting, but the underlying problem needs to be sorted out.  I think the error always happens after a "Bad Device Location" error message has been printed, which probably indicates the problem that sets up the chain or errors ending with the segfault.

dlmiles, 2024-11-11:
Thanks Tim for the feedback.

The following patch allows the `extresist` proc to complete.  There are a number (10) of "Bad Device" warnings over the design.  Investigating why will be for another day but at least this records the scenario via a project.

If a version of this patch was to be used, should the `extresist` proc return an error status, if 1 or more Bad Device warning(s) are emitted.  Then maybe an option to `extresist -ignore_bad_devices` to mitigate that error status into a success status, still reporting.
Then the error status is given it would emit a print/log entry informing the user of the number of bad devices being non-zero as the cause of the error and the availability of the ignore option to mitigate it.

This is just so scripts by default will continue to fail.  Previously they failed via SIGSEGV now they will fail gracefully via proc error status.

```
diff --git a/resis/ResJunct.c b/resis/ResJunct.c
index 96da39bb..7654b80c 100644
--- a/resis/ResJunct.c
+++ b/resis/ResJunct.c
@@ -55,6 +55,9 @@ ResNewSDDevice(tile, tp, xj, yj, direction, PendingList)
     tileJunk   *j;
 
     newnode = FALSE;
+    /* ResEachTile can call with CLIENTDEFAULT which is not a valid (tileJunk*) */
+    if (tp->ti_client == CLIENTDEFAULT)
+       return;  /* see Bad Device warnings during processing */
     j = (tileJunk *) tp->ti_client;
     resDev = j->deviceList;
     if ((j->sourceEdge & direction) != 0)
diff --git a/resis/ResMain.c b/resis/ResMain.c
index 1cbab2d0..d3f0090b 100644
--- a/resis/ResMain.c
+++ b/resis/ResMain.c
@@ -1725,7 +1725,10 @@ ResGetDevice(pt, type)
             return (((tileJunk *)tile->ti_client)->deviceList);
     }
     else if (TTMaskHasType(&ExtCurStyle->exts_deviceMask, TiGetType(tile)))
-        return (((tileJunk *)tile->ti_client)->deviceList);
+    {
+        if (tile->ti_client != CLIENTDEFAULT) /* see Bad Device warnings during processing */
+            return (((tileJunk *)tile->ti_client)->deviceList);
+    }
 
     return NULL;
 }
 ```

RTimothyEdwards, 2024-12-26:
I applied this patch to magic version 8.3.509.  I suggest leaving the issue open until the underlying "Bad Device" error is resolved.

dlmiles, 2025-02-26:
tileJunk==NULL

but also design triggers tg_Tdi=-1 which has an ASSERT at the top of ResWriteExtFile(), this log output logs the ASSERT was triggered but continues as if asserts disabled

Bad Device diagnostics:

```
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x24c2630 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x196bfa0 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x197f3e0 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x3caecd0 {.Cdownstream=0.000000, .Tdi=0.000000}]
Bad Device Location at 27403,21263 [why tj==NULL tt=scnmos(19) pNum=a(8)]
Bad Device Location at 27403,21555 [why tj==NULL tt=scpmoshvt(24) pNum=a(8)]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x3a18670 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x290e570 {.Cdownstream=0.000000, .Tdi=0.000000}]
Bad Device Location at 26115,21065 [why tj==NULL tt=scnmos(19) pNum=a(8)]
Bad Device Location at 26115,20719 [why tj==NULL tt=scpmoshvt(24) pNum=a(8)]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x2508590 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=(nil) {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=(nil) {.Cdownstream=0.000000, .Tdi=0.000000}]
Bad Device Location at 25011,21263 [why tj==NULL tt=scnmos(19) pNum=a(8)]
Bad Device Location at 25011,21555 [why tj==NULL tt=scpmoshvt(24) pNum=a(8)]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x1967920 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x1a3c560 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x19dfee0 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x19706e0 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 5.900000, totalcap = -1.000000, rc=0x24be7f0 {.Cdownstream=39.230850, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x19bb0a0 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x24b9e70 {.Cdownstream=0.000000, .Tdi=0.000000}]
Bad Device Location at 27127,21263 [why tj==NULL tt=scnmos(19) pNum=a(8)]
Bad Device Location at 27127,21555 [why tj==NULL tt=scpmoshvt(24) pNum=a(8)]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x19944a0 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=(nil) {.Cdownstream=0.000000, .Tdi=0.000000}]
ASSERT(ResRex.c:1864, gparams.rg_Tdi != -1, "ResWriteExtFile")
Bad Device Location at 28691,21065 [why tj==NULL tt=scnmos(19) pNum=a(8)]
Bad Device Location at 28691,20719 [why tj==NULL tt=scpmoshvt(24) pNum=a(8)]
Adding  input2.X; Tnew = 0.12ns, Told = 0.00ns
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 2.200000, totalcap = -1.000000, rc=0x356c670 {.Cdownstream=96.733208, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x3d594d0 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=0x3aaf8a0 {.Cdownstream=0.000000, .Tdi=0.000000}]
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 10.900000, totalcap = -1.000000, rc=(nil) {.Cdownstream=0.000000, .Tdi=0.000000}]
ASSERT(ResRex.c:1864, gparams.rg_Tdi != -1, "ResWriteExtFile")
goodies->rg_Tdi = -1, why [goodies->rg_nodecap == 0.000000, totalcap = -1.000000, rc=(nil) {.Cdownstream=0.000000, .Tdi=0.000000}]
ASSERT(ResRex.c:1864, gparams.rg_Tdi != -1, "ResWriteExtFile")

```