RTimothyEdwards/magic #354: Fatai IO Error 104

MostlyKIGuess, 2024-11-12:
[image]

- I tried opening with -noconsole but no luck sadly, and it just crashed while I was extending metal2, it happened out of the blue.
- Not sure how to reconstruct.
### OS Information:
- Arch Linux
- Magic Version: Latest

MostlyKIGuess, 2024-11-12:
I was running on Wayland and had TCL , TK lib installed.

RTimothyEdwards, 2024-12-26:
The issue here, `couldn't connect to display ":1"`, generally means that there isn't an X11 server running.  It seems very weird to me that it would happen in the middle of running the program.  It's like the X server crashed or something.  You might want to try running magic with `-d XR` (requires installing cairo graphics library) which would avoid making low-level X11 calls.

Sorry about the delay in responding.  I had to get through a tape-out and into the holiday before I had enough time to respond to emails and issues from the last couple of months.

RTimothyEdwards, 2024-12-26:
My best guess is that this is probably a bug in the X11 implementation in Wayland, although I cannot discount the possibility of a bug in magic's X11 calls, especially around the interactive wiring and copying pixmaps rapidly between the application and backing store.  Hopefully using cairo or OpenGL graphics avoids the problem altogether.

MostlyKIGuess, 2024-12-27:
> My best guess is that this is probably a bug in the X11 implementation in Wayland, although I cannot discount the possibility of a bug in magic's X11 calls, especially around the interactive wiring [...]

Yes it somehow is connected to handling of the tech files btw, for some reason using the same tech file updating via GUI would not consider it and not update the corresponding metal similar to terminal as well, I would run the command twice or thrice and it would work giving proper DRC checks and correct metals for the files. Would you happen to know about this issue? Perhaps if there's any lead I would love to work on it