RTimothyEdwards/magic #355: Compilation Error on MacOS: typedef redefinition of ClientData

Qsionc, 2025-01-01:
Hi and happy new year,

I am trying to build magic from source on MacOS, following steps from INSTALL_MacOS.md, trying my own small tweaks and nothing seems to work. I did not have the time to resolve this, but either there is something missing with preprocessor definitions or name conflicts, or something entirely else.

It is the only error generated for me as of now, i have every prerequisite installed and properly recognized. 

Error text:
```shell
In file included from CMWcmmnds.c:27:
../utils/magic.h:63:21: error: typedef redefinition with different types ('pointertype' (aka 'unsigned long') vs 'void *')
   63 | typedef pointertype ClientData;
      |                     ^
/opt/homebrew/Cellar/tcl-tk/9.0.1/include/tcl-tk/tcl.h:233:15: note: previous definition is here
  233 | typedef void *ClientData;
  ```

RTimothyEdwards, 2025-01-01:
I have not come across this one before.  It is a throwback to the fact that magic was written by John Ousterhout, who went on to write Tcl/Tk, and magic and Tcl share a bunch of code as well as variable names.

My first suggestion would be to just add `#define _CLIENTDATA` at `utils/magic.h` line 61, in which case the definition from Tcl would be used.  That might only serve to kick the errors down the road a little bit, though.

Qsionc, 2025-01-01:
I did add to CFLAGS `-D_CLIENTDATA` but it created additional errors coming from incompatibility of `void*` and `unsigned long int` casts along the way in some places. 

I might also try and build magic without tcl/tk entirely, I will try that. But, i will also try to debug and fix it in free time (maybe create PR at some point, if i am fast enough). Thanks for insights on the matter ;)

RTimothyEdwards, 2025-01-01:
I don't think `-D_CLIENTDATA` will work because the Tcl code _also_ has an `#ifdef _CLIENTDATA` around it (more proof that Tcl/Tk started from magic code).

dlmiles, 2025-01-01:
Maybe try with TCL 8.6+ instead of TCL9.  By installing a version 8 of tk/tcl on MacOS.

This is a known matter recorded in https://github.com/RTimothyEdwards/magic/pull/351 that the type was changed between TCL8 and TCL9 and maybe an GNU ./configure style configuration can resolve it.

Maybe once #351 is closed then TCL9 can be expected to work.

Qsionc, 2025-01-01:
It indeed worked! I used pre-built tcl-tk@8 package from homebrew, passed it to configure script with proper options and it built!

Now i have to understand why was there a segfault when starting the program, but i will get to it... ;) Thank you, the build issue is resolved!

closed 2025-01-08 by Qsionc

shaunplee, 2025-01-14:
> Now i have to understand why was there a segfault when starting the program, but i have to make some script-magic to add tcl-tk@8/bin to path for magic without breaking my system ;) Thank you, the build issue is resolved!

@Qsionc you might be interested in: https://github.com/RTimothyEdwards/magic/issues/280