RTimothyEdwards/magic #367: segfault when running caravel/caravan composite.py on compressed files

d-m-bailey, 2025-02-15:
Version 8.3.519

See https://github.com/efabless/caravel/issues/564

Is magic supposed to implicitly recognize and open compressed mag files similar to gds file references?

RTimothyEdwards, 2025-02-15:
Yes.  But unlike GDS, there is no way in magic to _create_ a compressed .mag file.

d-m-bailey, 2025-02-15:
Here's the script `../../mag/compose_final.tcl` (note that the make targets `set_user_id`, `gpio_defaults`, `truck` and `generate_fill` have already been executed).
```
#!/bin/env wish
drc off
crashbackups stop
locking disable
addpath /home/2504/caravel_user_project_analog/mgmt_core_wrapper/mag
random seed 621019138
load caravan -dereference
property GDS_FILE /home/2504/caravel_user_project_analog/gds/caravan.gds
property GDS_START 0
select top cell
set bbox [box values]
load caravel_25040002_fill_pattern -silent
snap internal
box values {*}$bbox
paint comment
property GDS_FILE /home/2504/caravel_user_project_analog/gds/caravel_25040002_fill_pattern.gds
property GDS_START 0
property FIXED_BBOX "$bbox"
load caravel_25040002 -silent
box values 0 0 0 0
box position 6um 6um
getcell caravan child 0 0
getcell caravel_25040002_fill_pattern child 0 0
box position 0 0
getcell advSeal_6um_gen
puts stdout "Writing final GDS. . . "
flush stdout
gds undefined allow
cif *hier write disable
gds write /home/2504/caravel_user_project_analog/gds/caravel_25040002.gds
quit -noprompt
```

Here's the debug trace
```
~/2504/caravel_user_project_analog/caravel/mag$ gdb /home/local/lib/magic/tcl/magicdnull
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
…
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/kanobailey/local/lib/magic/tcl/magicdnull...done.
(gdb) run -nowrapper -rcfile /home/2504/caravel_user_project_analog/dependencies/pdks/sky130A/libs.tech/magic/sky130A.magicrc ../../mag/compose_final.tcl
Starting program: /home/local/lib/magic/tcl/magicdnull -nowrapper -rcfile /home/2504/caravel_user_project_analog/dependencies/pdks/sky130A/libs.tech/magic/sky130A.magicrc ../../mag/compose_final.tcl
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Magic 8.3 revision 519 - Compiled on Thu 13 Feb 2025 11:30:51 AM PST.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using TrueColor, VisualID 0x21 depth 24
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
Input style sky130(): scaleFactor=2, multiplier=2
The following types are not handled by extraction and will be treated as non-electrical types:
    ubm 
Scaled tech values by 2 / 1 to match internal grid scaling
Loading sky130A Device Generator Menu ...
Loading "../../mag/compose_final.tcl" from command line.
Cell caravan read from current working directory
Processing timestamp mismatches.
Cell user_id_textblock read from current working directory
Scaled magic input cell user_id_textblock geometry by factor of 2
Cell chip_io_alt read from current working directory
Cell open_source read from current working directory
Scaled magic input cell open_source geometry by factor of 2
Cell copyright_block_a read from current working directory
...
[... 74 lines cut ...]
(gdb) l 1160
1155        /* If not an array element, life is much simpler */
1156        if (use->cu_xlo == use->cu_xhi && use->cu_ylo == use->cu_yhi)
1157        {
1158            newScx.scx_x = use->cu_xlo, newScx.scx_y = use->cu_yhi;
1159            if (SigInterruptPending) return 1;
1160            GEOINVERTTRANS(&use->cu_transform, &tinv);
1161            GeoTransTrans(&use->cu_transform, &scx->scx_trans,
1162                                    &newScx.scx_trans);
1163            GEOTRANSRECT(&tinv, &scx->scx_area, &newScx.scx_area);
1164            if ((*fp->tf_func)(&newScx, fp->tf_arg) == 1)
(gdb) p *fp
$2 = {tf_func = 0x7ffff67cf403 <dbReadAreaFunc>, tf_arg = 0x0, tf_mask = 0xaaffffc8e0, tf_xmask = 9765332, tf_planes = -17179869184, tf_dinfo = 170, tf_flags = 17 '\021', 
  tf_tpath = 0x121000001d4}
(gdb) p *scx
$3 = {scx_use = 0x555557d3c9d0, scx_x = 0, scx_y = 0, scx_area = {r_ll = {p_x = -21341, p_y = -1200}, r_ur = {p_x = 738940, p_y = 1038800}}, scx_trans = {t_a = 1, t_b = 0, 
    t_c = 1200, t_d = 0, t_e = 1, t_f = 1200}}
(gdb) p *(scx->scx_use)
$4 = {cu_bpLinks = {0x0, 0x0, 0x555556e0f668}, cu_bbox = {r_ll = {p_x = 1200, p_y = 1200}, r_ur = {p_x = 718800, p_y = 1038800}}, cu_extended = {r_ll = {p_x = -20141, 
      p_y = 1200}, r_ur = {p_x = 740140, p_y = 1038800}}, cu_expandMask = 0, cu_flags = 0 '\000', cu_transform = {t_a = 1, t_b = 0, t_c = 1200, t_d = 0, t_e = 1, t_f = 1200}, 
  cu_id = 0x55555765e4b0 "caravan_0", cu_array = {ar_xlo = 0, ar_xhi = 0, ar_ylo = 0, ar_yhi = 0, ar_xsep = 0, ar_ysep = 0}, cu_def = 0x5555576f0820, cu_nextuse = 0x55555767c530, 
  cu_parent = 0x555556e9a3c0, cu_client = 0xc000000000000004}
does this help?
fp->tf_arg is being passed as a null pointer.
```

d-m-bailey, 2025-02-15:
I think this happens when the `caravel/mag/caravan_core.mag` file is compressed.

d-m-bailey, 2025-02-15:
The above occurs when `MAGTYPE=mag` and `caravel/mag/caravan_core.mag` is compressed.

There's different error when `MAGTYPE=maglef` (`caravel/maglef/caravan_core.mag` is not compressed).

```
...
Reading "sky130_fd_sc_hvl__fill_1".
Reading "sky130_fd_sc_hvl__fill_2".
Reading "sky130_fd_sc_hvl__lsbufhv2lv_1".
Reading "xres_buf".
Reading "sky130_fd_sc_hd__einvp_1".
Reading "sky130_fd_sc_hd__clkinv_1".
free(): invalid next size (fast)
```

Here's the gdb results
```
Reading "xres_buf".
Reading "sky130_fd_sc_hd__einvp_1".
Reading "sky130_fd_sc_hd__clkinv_1".
free(): invalid next size (fast)

Program received signal SIGABRT, Aborted.
0x00007ffff7a278eb in raise () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) bt
#0  0x00007ffff7a278eb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff7a12535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff7a69648 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#3  0x00007ffff7a6fd6a in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#4  0x00007ffff7a71646 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#5  0x00007ffff68d4b59 in freeMagic (cp=0x55555c1493d0) at malloc.c:146
#6  0x00007ffff68faa7d in calmaDumpStructure (def=0x555556c53980, outf=0x5555558b27a0, calmaDefHash=0x7fffffffc810, 
    filename=0x7ffff6a1dfe0 <realName.15786> "/home/kanobailey/2504/caravel_user_project_analog/gds/caravan.gds") at CalmaWrite.c:572
#7  0x00007ffff68fb291 in calmaFullDump (def=0x555556c53980, fi=0x555557653010, outf=0x5555558b27a0, 
    filename=0x7ffff6a1dfe0 <realName.15786> "/home/kanobailey/2504/caravel_user_project_analog/gds/caravan.gds") at CalmaWrite.c:772
#8  0x00007ffff68fb858 in calmaProcessDef (def=0x555556c53980, outf=0x5555558b27a0, do_library=0 '\000') at CalmaWrite.c:1022
#9  0x00007ffff68fb3a0 in calmaProcessUse (use=0x55555b71ad10, outf=0x5555558b27a0) at CalmaWrite.c:854
#10 0x00007ffff67ca209 in dbEnumFunc (use=0x55555b71ad10, fp=0x7fffffffca20) at DBcellsrch.c:1271
#11 0x00007ffff67c8134 in DBSrCellPlaneArea (plane=0x555557addbb0, rect=0x7ffff6a0bd40 <TiPlaneRect>, func=0x7ffff67ca1cd <dbEnumFunc>, arg=0x7fffffffca20) at DBcellsrch.c:97
#12 0x00007ffff67ca1bb in DBCellEnum (cellDef=0x555557add830, func=0x7ffff68fb374 <calmaProcessUse>, cdarg=0x5555558b27a0) at DBcellsrch.c:1239
#13 0x00007ffff68fb651 in calmaProcessDef (def=0x555557add830, outf=0x5555558b27a0, do_library=0 '\000') at CalmaWrite.c:955
#14 0x00007ffff68fa03d in CalmaWrite (rootDef=0x555557add830, f=0x5555558b27a0) at CalmaWrite.c:343
#15 0x00007ffff679c1c7 in CmdCalma (w=0x5555558483c0, cmd=0x55555751d5a0) at CmdCD.c:1051
#16 0x00007ffff688ec62 in WindExecute (w=0x5555558483c0, rc=0x5555558a7590, cmd=0x55555751d5a0) at windMain.c:419
#17 0x00007ffff67fed37 in DBWcommands (w=0x5555558483c0, cmd=0x55555751d5a0) at DBWprocs.c:688
#18 0x00007ffff688b7f3 in WindSendCommand (w=0x5555558483c0, cmd=0x55555751d5a0, quiet=1 '\001') at windSend.c:263
#19 0x00007ffff6883c60 in TxTclDispatch (clientData=0x0, argc=3, argv=0x555555574830, quiet=1 '\001') at txCommands.c:1391
#20 0x00007ffff69a8fd8 in _tcl_dispatch (clientData=0x0, interp=0x555555566980, argc=3, argv=0x555555574830) at tclmagic.c:430
#21 0x00007ffff7c0d1eb in TclInvokeStringCommand () from /lib/x86_64-linux-gnu/libtcl8.6.so
#22 0x00007ffff7c0efb7 in TclNRRunCallbacks () from /lib/x86_64-linux-gnu/libtcl8.6.so
#23 0x00007ffff7c103af in ?? () from /lib/x86_64-linux-gnu/libtcl8.6.so
#24 0x00007ffff7cc93f8 in Tcl_FSEvalFileEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#25 0x00007ffff7cc7d57 in Tcl_EvalFile () from /lib/x86_64-linux-gnu/libtcl8.6.so
#26 0x00007ffff68d4911 in mainInitFinal () at main.c:1206
#27 0x00007ffff69a9a96 in _magic_startup (clientData=0x0, interp=0x555555566980, argc=1, argv=0x5555555743f0) at tclmagic.c:771
#28 0x00007ffff7c0d1eb in TclInvokeStringCommand () from /lib/x86_64-linux-gnu/libtcl8.6.so
#29 0x00007ffff7c0efb7 in TclNRRunCallbacks () from /lib/x86_64-linux-gnu/libtcl8.6.so
#30 0x00007ffff7c103af in ?? () from /lib/x86_64-linux-gnu/libtcl8.6.so
--Type <RET> for more, q to quit, c to continue without paging-- 
#31 0x00007ffff7cc93f8 in Tcl_FSEvalFileEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#32 0x00007ffff7cc7d57 in Tcl_EvalFile () from /lib/x86_64-linux-gnu/libtcl8.6.so
#33 0x00007ffff7cd0f6e in Tcl_SourceRCFile () from /lib/x86_64-linux-gnu/libtcl8.6.so
#34 0x00007ffff7cd14bd in Tcl_MainEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#35 0x000055555555522d in main (argc=5, argv=0x7fffffffdb98) at magicdnull.c:49
(gdb) up
#1  0x00007ffff7a12535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) up
#2  0x00007ffff7a69648 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) up
#3  0x00007ffff7a6fd6a in ?? () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) up
#4  0x00007ffff7a71646 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) up
#5  0x00007ffff68d4b59 in freeMagic (cp=0x55555c1493d0) at malloc.c:146
146             FreeRoutine(freeDelayedItem);
(gdb) l
141             TxError("freeMagic called with NULL argument.\n");
142         if (freeDelayedItem)
143         {
144             /* fprintf(stderr, "freed 0x%x\n", freeDelayedItem); fflush(stderr); */
145
146             FreeRoutine(freeDelayedItem);
147         }
148         freeDelayedItem=cp;
149     }
150
(gdb) p freeDelayedItem
$1 = 0x55555c149940 "~/2504/caravel_user_project/dependencies/pdks/sky130A/libs.ref/sky130_fd_sc_hd/maglef/sky130_fd_sc_hd__clkinv_1"
```

This may be because the new pdk causes a file pointer overflow and I added `locking disable` to the script.

Checking now...

RTimothyEdwards, 2025-02-15:
@d-m-bailey :  Apparently the `gds undefined allow` command option has not been thoroughly vetted because it causes a NULL value to be passed for `err_ptr` and the NULL value isn't being checked by some routines.  Easy enough to fix.

d-m-bailey, 2025-02-15:
@RTimothyEdwards Thanks for the quick response.

Removing the locking disable and bumping the ulimit for open files to 2048 still gives an error for the maglef version.

RTimothyEdwards, 2025-02-15:
@d-m-bailey :  The crash condition has been fixed.  Note, however, that the underlying problem is that `gds undefined allow` was set, and an undefined cell was found.  But `gds undefined allow` should never be issued for a tapeout---even if the code didn't crash, the GDS would be bad and would be rejected by the foundry.  The issue at the bottom is that there was a cell that was not found when trying to write GDS.  I recommend removing/commenting `gds undefined allow` and re-running to figure out what cell was the root of the problem (doesn't require using the updated version of magic).

RTimothyEdwards, 2025-02-15:
@d-m-bailey :  The ulimit error is very rare and I have only ever seen it when using the gf180mcu PDK.

d-m-bailey, 2025-02-15:
@RTimothyEdwards 

> I recommend removing/commenting `gds undefined all` and re-running to figure out what cell was the root of the problem 

Same error, unfortunately.

> The ulimit error is very rare and I have only ever seen it when using the gf180mcu PDK.

I doubt that it's a question of rarity, but rather the number of files that have to be open simultaneously. There have apparently been additions to the pdk (fill cells, etc.) that have caused the total to exceed 1024.

I'd previously stated

> There's different error when `MAGTYPE=maglef` (`caravel/maglef/caravan_core.mag` is not compressed).

But this is incorrect. The script is run in the `caravel/mag` directory so the current directory `caravel/mag/caravan_core.mag` file is read (and fails with an error if the mag file is compressed).

With `gds undefined all` not set, this is the `stderr` output
```
Cell caravan_core couldn't be read
Failure to read entire subtree of the cell.
Failed on cell caravan_core.
```

RTimothyEdwards, 2025-02-16:
@d-m-bailey :  Is it really the same error, though?  I am not especially surprised if it still crashes, but if so it should be crashing in a different place and hopefully closer to the actual problem.

d-m-bailey, 2025-02-16:
@RTimothyEdwards As far as I could tell, it was the exact same place. I'll recheck though.

d-m-bailey, 2025-02-16:
These are the results for magic `8.3.519`

This is the script that I execute in the `caravel/mag` directory.
Note that there are these changes from the original script:
1. added `crashbackups stop` 
1. added `locking disable` 
2. added `addpath` for `mgmt_core_wrapper`
3. removed `gds undefined allow`
```
#!/bin/env wish
drc off
crashbackups stop
locking disable
addpath /home/2504/caravel_user_project_analog/mgmt_core_wrapper/mag
random seed 621019138
load caravan -dereference
property GDS_FILE /home/2504/caravel_user_project_analog/gds/caravan.gds
property GDS_START 0
select top cell
set bbox [box values]
load caravel_25040002_fill_pattern -silent
snap internal
box values {*}$bbox
paint comment
property GDS_FILE /home/2504/caravel_user_project_analog/gds/caravel_25040002_fill_pattern.gds
property GDS_START 0
property FIXED_BBOX "$bbox"
load caravel_25040002 -silent
box values 0 0 0 0
box position 6um 6um
getcell caravan child 0 0
getcell caravel_25040002_fill_pattern child 0 0
box position 0 0
getcell advSeal_6um_gen
puts stdout "Writing final GDS. . . "
flush stdout
#gds undefined allow
cif *hier write disable
gds write /home/2504/caravel_user_project_analog/gds/caravel_25040002.gds
quit -noprompt
```

With `caravan_core.mag` uncompressed, the following completes normally and creates the final gds.
```
$ MAGTYPE=mag magic -dnull -noc -rcfile $PDK_ROOT/$PDK/libs.tech/magic/sky130A.magicrc compose_final.tcl 
$PDK_ROOT/$PDK/libs.tech/magic/sky130A.tech: line 6087: section (none):
        Unrecognized layer (type) name "res0p69"
Warning:  Parent cell lists instance of "EF_fill_4_8" at bad file path $PDK_ROOT/$PDK/libs.ref/sky130_fd_sc_hd/mag/EF_fill_4_8.mag.
The cell exists in the search paths at EF_fill_4_8.mag.
The discovered version will be used.
```
The commands below all result incomplete gds data.

With `caravan_core.mag` compressed, this is the result:
```
$ MAGTYPE=mag magic -dnull -noc -rcfile $PDK_ROOT/$PDK/libs.tech/magic/sky130A.magicrc compose_final.tcl
Cell caravan_core couldn't be read
Failure to read entire subtree of the cell.
Failed on cell caravan_core.
I/O error in writing file /home/kanobailey/2504/caravel_user_project_analog/gds/caravel_25040002.gds.
File may be incompletely written.
```

With `caravan_core.mag` uncompressed and the `MAGTYPE=maglef` (as in original script), this is the result:
```
$ MAGTYPE=maglef magic -dnull -noc -rcfile $PDK_ROOT/$PDK/libs.tech/magic/sky130A.magicrc compose_final.tcl 
Duplicate cell in RAM128:  Instance of cell sky130_fd_sc_hd__clkbuf_16 is from path /home/kanobailey/2504/caravel_user_project_analog/dependencies/pdks/sky130A/libs.ref/sky130_fd_sc_hd/mag but cell was previously read from /home/kanobailey/2504/caravel_user_project_analog/dependencies/pdks/sky130A/libs.ref/sky130_fd_sc_hd/maglef.
Cell name conflict:  Renaming original cell to sky130_fd_sc_hd__clkbuf_16__0.
Duplicate cell in RAM128:  Instance of cell sky130_fd_sc_hd__clkbuf_4 is from path /home/kanobailey/2504/caravel_user_project_analog/dependencies/pdks/sky130A/libs.ref/sky130_fd_sc_hd/mag but cell was previously read from /home/kanobailey/2504/caravel_user_project_analog/dependencies/pdks/sky130A/libs.ref/sky130_fd_sc_hd/maglef.
Cell name conflict:  Renaming original cell to sky130_fd_sc_hd__clkbuf_4__0.
/home/kanobailey/2504/caravel_user_project_analog/dependencies/pdks/sky130A/libs.tech/magic/sky130A.tech: line 6087: section (none):
        Unrecognized layer (type) name "res0p69"
Warning:  Parent cell lists instance of "EF_fill_4_8" at bad file path /home/kanobailey/2504/caravel_user_project_analog/dependencies/pdks/sky130A/libs.ref/sky130_fd_sc_hd/mag/EF_fill_4_8.mag.
The cell exists in the search paths at EF_fill_4_8.mag.
The discovered version will be used.
free(): invalid next size (fast)
Aborted
```
The addition of the `addpath` appears to cause duplicate cells that result in the `free` error.

The original script did not include the `mgmt_core_wrapper` path and removing the `addpath` line in the current script gives this result:
```
$ MAGTYPE=maglef magic -dnull -noc -rcfile  $PDK_ROOT/$PDK/libs.tech/magic/sky130A.magicrc compose_final.noram.tcl 
Cell RAM128 couldn't be read
Failure to read entire subtree of the cell.
Failed on cell RAM128.
I/O error in writing file /home/kanobailey/2504/caravel_user_project_analog/gds/caravel_25040002.gds.
File may be incompletely written.
```
Since the `gds undefined allow` line has been removed, gds write fails. The actual mag RAM file is not used in the final data, since the gds should override it. Maybe that's why the `gds undefined allow` was in the original script.

Adding `gds undefined allow`, results in the original segmentation error.
```
$ MAGTYPE=maglef magic -dnull -noc -rcfile  $PDK_ROOT/$PDK/libs.tech/magic/sky130A.magicrc compose_final.undef-ok.tcl 
Cell RAM128 couldn't be read
Segmentation fault
```

I've modified the script to run without errors (it uncompresses the `caravan_core.mag` file and uses `MAGTYPE=mag`).
It would be great if magic would
1. Read compressed sub-cells natively (in this case, `caravan_core.mag.gz`)
2. When executing gds write, if there is a gds override, ignore the hierarchy defined in the mag files

dlmiles, 2025-02-16:
> Removing the locking disable and bumping the ulimit for open files to 2048 still gives an error for the maglef version.

FWIW I have in pipeline numerous CodeQL resolutions for leaked file-descriptors, not sure if that will make any difference to this issue, I guess it depends on your tcl command/loops.  Most of the places seemed obscure probably lesser used parts of magic, but the FILE_LOCKS use of `cellDef->cd_fd = -2;` is something I have not got to understand yet (the WIP commit touches this area).  Maybe this should be a reference counted handover of the FD closure management a kind of `flock_close()` so at least the last user will make a close syscall ?

One concern with raising the ulimit is the use of `fd_set` in the project for some parts of the code base (which is effectively a compile time limit on the max FD number that can be used around fd_set).  This maybe restricted to console/X11-socket setup that is always setup early in the process lifetime, before the active FD count gets near the fd_set limit.