RTimothyEdwards/magic #368: Layout of Analog Components

Camisad0, 2025-02-15:
Hi, do you have any tutorial or guidelines on how to layout passive devices like caps and resistors? Also, for PNP transistors? Thanks!

RTimothyEdwards, 2025-02-15:
What technology?  I think the best way is to install a full-featured technology like sky130 and look at how the generated devices (a.k.a. pcells) are drawn.  Every technology has specific requirements for resistors and capacitors.

Camisad0, 2025-02-20:
Hi Timothy, I'm using Sky130 tech. I'm just wondering if there are guidelines in laying out analog passive elements (i.e. resistors, caps, pnp transistors). Context: I'm laying out for an Op-Amp based BGR circuit.

RTimothyEdwards, 2025-02-20:
Use the "import SPICE" menu option in the sky130 PDK for magic to convert schematic (netlist) devices into something reasonable for analog layout.  After that, you can modify the devices for layout-specific options.  If you have questions about best layout practices, I suggest the #analog-design channel on the open-source-silicon Slack workspace.