RTimothyEdwards/magic #397: Labels with newline break saved .mag files

urish, 2025-03-20:
Given the following GDS file:

[tt_ihp_wrapper.gds.zip](https://github.com/user-attachments/files/19375119/tt_ihp_wrapper.gds.zip)

When I try to convert it to a bunch of a .mag files, one of the cells results in an invalid .mag format.

This is how I convert:

```
gds readonly true
gds rescale false
gds read tt_ihp_wrapper.gds
writeall force tt_ihp_wrapper
writeall modified
```

After running the above commands, sealring.mag has an invalid format and fails to load in magic, due to the following lines in the saved sealring.mag file:

```
<< labels >>
flabel comment s 500 500 500 500 2 FreeSans 5000 0 0 0 Device registration size: x=5400.0 um ; y=6450.0 um
Calculated area: 3.5e-05 sq mm
```

Looks like the newline in the text coming from the GDS file somehow breaks magic.

This is what the offending sealring cell / text object looks like in klayout, where you can see the newline in the text:

[image]

RTimothyEdwards, 2025-03-21:
I have never known a newline to be a valid GDS character.  Even "permissive" GDS doesn't allow control characters (any character below 32).

RTimothyEdwards, 2025-03-21:
However, since magic just does a read() from the file, it picks up whatever is in the string and doesn't try to map it the same way it maps GDS output.  But it should be easy enough to fix;  it just requires escaping control characters that it writes to the .mag file output.

dlmiles, 2025-03-21:
```
00289b60  xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx 56  |...............V|
00289b70  19 06 44 65 76 69 63 65  20 72 65 67 69 73 74 72  |..Device registr|
00289b80  61 74 69 6f 6e 20 73 69  7a 65 3a 20 78 3d 35 34  |ation size: x=54|
00289b90  30 30 2e 30 20 75 6d 20  3b 20 79 3d 36 34 35 30  |00.0 um ; y=6450|
00289ba0  2e 30 20 75 6d 0a 43 61  6c 63 75 6c 61 74 65 64  |.0 um.Calculated|
00289bb0  20 61 72 65 61 3a 20 33  2e 35 65 2d 30 35 20 73  | area: 3.5e-05 s|
00289bc0  71 20 6d 6d 00 xx xx xx  xx xx xx xx xx xx xx xx  |q mm............|
```

Any preference on the escape method ?  `\x0a` and `\\` and replace all chars < 32 and >= 127 ?

magic needs to be able to load files it generated itself

RTimothyEdwards, 2025-03-21:
@dlmiles :  I think that's sufficient.  I question whether what klayout is doing is valid, because as I said, it does not appear to be compatible with the GDS spec.  But at the same time, foundries may not care, because text isn't a mask layer.  Dealing with the carriage return is not something that I would really expect to need to handle in magic, but when printing a label it will need to do _something_.  Also it may be prudent to check that any problematic character only appears in a non-electrically-connected layer, otherwise if it is in a net name (and that includes spaces, which I think _are_ checked) it will screw up the SPICE simulator or LVS tool.

dlmiles, 2025-03-21:
Maybe a warning is good enough with the prudent check part ?

The SPICE/LVS emitters should also encode unexpected characters in a way compatible maybe by just replacing each with an underscore `_` or `_0a` style encoding ?  This only seems a problem if it generates a duplicate name/label because another item already has an original claim to the new name.  Then maybe it can start suffixing to make unique.

In which case maybe data emitters need to resolve all the (potentially renamed) name/label first, before and emitter starts to generate data it has a consistent view of the label given in the target namespace (SPICE/LVS).  and maybe a mapping information textfile to kick out if the information is useful to help identify a rename occurred and to resolve names.

dlmiles, 2025-06-29:
I modified the sample script to:

```
drc off
gds readonly true
gds rescale false
gds read tt_ihp_wrapper.gds
writeall force tt_ihp_wrapper
#select top cell
cellname rename "(UNNAMED)" top_tt_ihp_wrapper
#select clear
writeall modified
```

```
$ grep -B1 -A1 label *.mag | fgrep -B1 -A1 "\\x"
sealring.mag:<< labels >>
sealring.mag:flabel comment s 500 500 500 500 2 FreeSans 5000 0 0 0 Device registration size: x=5400.0 um ; y=6450.0 um\x0aCalculated area: 3.5e-05 sq mm
sealring.mag-<< properties >>
```

The change (not get in PR pending feedback here and cleanup) only affects the save/load routines for the `*.mag` format, with `flabel` and `rlabel` parser/formatter for the `text` field.  So does this mitigate the concern with the `cellname` having non-strict-ASCII printables.  The `text` field seems designed to be human readable and not an identifier.

The question is how would an unexpected character into a cellname/symbol name ?

It does seem possible to for Tcl to attempt to set cellname:

`cellname rename "(UNNAMED)" "broken_\nlabel"` ends up as a file `broken_` no `*.mag` extension, but otherwise consistent.

It does seem to keep it with such a name:
```
% cellname list celldef "broken_\nlabel"
{broken_
label}
% cellname list celldef "broken_"
0
% cellname list celldef "broken_\nlabelZ"
0
```

dlmiles, 2025-06-30:
FYI  Looks like 2 top cells.

```
% cellname list window
(UNNAMED)
% cellname list topcells
(UNNAMED) tt_ihp_wrapper
% cellname list allcells
(UNNAMED) bondpad_70x70 bondpad_CDNS_701964819630 E5_sg13g2_and2_1 E5_sg13g2_and2_2 E5_sg13g2_and3_1 E5_sg13g2_antennanp E5_sg13g2_buf_2 E5_sg13g2_buf_8 E5_sg13g2_decap_4 E5_sg13g2_decap_8 E5_sg13g2_ebufn_4 E5_sg13g2_ebufn_8 E5_sg13g2_fill_1 E5_sg13g2_fill_2 E5_sg13g2_inv_1 E5_sg13g2_inv_2 E5_sg13g2_inv_8 E5_sg13g2_mux4_1 E5_sg13g2_nand2_1 E5_sg13g2_nand2b_1 E5_sg13g2_nand4_1 E5_sg13g2_nor2_1 E5_sg13g2_nor2b_1 E5_sg13g2_nor4_1 E5_sg13g2_o21ai_1 E5_sg13g2_tiehi E5_sg13g2_tielo E5_sg13g2_xor2_1 LL_sg13g2_and2_2 LL_sg13g2_antennanp LL_sg13g2_buf_2 LL_sg13g2_buf_8 LL_sg13g2_decap_4 LL_sg13g2_decap_8 LL_sg13g2_dfrbp_2 LL_sg13g2_ebufn_4 LL_sg13g2_fill_1 LL_sg13g2_fill_2 LL_sg13g2_inv_1 LL_sg13g2_inv_2 LL_sg13g2_inv_8 LL_sg13g2_mux2_1 LL_sg13g2_tiehi LL_sg13g2_tielo OD_sg13g2_a21oi_1 OD_sg13g2_a21oi_2 OD_sg13g2_a22oi_1 OD_sg13g2_and2_1 OD_sg13g2_and2_2 OD_sg13g2_and3_1 OD_sg13g2_and3_2 OD_sg13g2_buf_1 OD_sg13g2_buf_2 OD_sg13g2_buf_4 OD_sg13g2_decap_4 OD_sg13g2_decap_8 OD_sg13g2_dfrbp_1 OD_sg13g2_dlygate4sd3_1 OD_sg13g2_fill_1 OD_sg13g2_fill_2 OD_sg13g2_inv_1 OD_sg13g2_inv_4 OD_sg13g2_nand2b_1 OD_sg13g2_nand4_1 OD_sg13g2_nor2_1 OD_sg13g2_tiehi OD_sg13g2_xnor2_1 OD_sg13g2_xor2_1 sealring sg13g2_Clamp_N2N2D sg13g2_Clamp_N43N43D4R sg13g2_Clamp_P2N2D sg13g2_Corner sg13g2_DCNDiode sg13g2_DCPDiode sg13g2_Filler200 sg13g2_Filler400 sg13g2_Filler1000 sg13g2_Filler2000 sg13g2_Filler4000 sg13g2_Filler10000 sg13g2_GateDecode sg13g2_GateLevelUpInv sg13g2_GuardRing_N1324W456HTF sg13g2_GuardRing_N7276W2716HFF sg13g2_GuardRing_N9472W2216HTT sg13g2_GuardRing_N15280W3132HTF sg13g2_GuardRing_N16000W1980HFF sg13g2_GuardRing_N16000W4884HFF sg13g2_GuardRing_N16000W6624HFF sg13g2_GuardRing_P456W948HFF sg13g2_GuardRing_P576W948HFF sg13g2_GuardRing_P7276W2716HFF sg13g2_GuardRing_P15280W1260HFF sg13g2_GuardRing_P15280W4164HFF sg13g2_GuardRing_P16000W3852HFF sg13g2_GuardRing_P16000W4466HFT sg13g2_io_inv_x1 sg13g2_io_nand2_x1 sg13g2_io_nor2_x1 sg13g2_io_tie sg13g2_IOPadInOut4mA sg13g2_IOPadIn sg13g2_IOPadIOVdd sg13g2_IOPadIOVss sg13g2_IOPadOut4mA sg13g2_IOPadVdd sg13g2_IOPadVss sg13g2_LevelDown sg13g2_LevelUpInv sg13g2_LevelUp sg13g2_RCClampInverter sg13g2_RCClampResistor sg13g2_SecondaryProtection tt_ctrl tt_ihp_wrapper tt_mux tt_um_chip_rom tt_um_factory_test W5_sg13g2_a21o_1 W5_sg13g2_a21o_2 W5_sg13g2_a21oi_1 W5_sg13g2_a21oi_2 W5_sg13g2_a22oi_1 W5_sg13g2_a221oi_1 W5_sg13g2_and2_1 W5_sg13g2_and3_1 W5_sg13g2_and3_2 W5_sg13g2_and4_1 W5_sg13g2_buf_1 W5_sg13g2_buf_2 W5_sg13g2_buf_4 W5_sg13g2_decap_4 W5_sg13g2_decap_8 W5_sg13g2_dfrbp_1 W5_sg13g2_dlygate4sd3_1 W5_sg13g2_fill_1 W5_sg13g2_fill_2 W5_sg13g2_inv_1 W5_sg13g2_inv_2 W5_sg13g2_mux2_1 W5_sg13g2_mux2_2 W5_sg13g2_nand2_1 W5_sg13g2_nand2_2 W5_sg13g2_nand2b_1 W5_sg13g2_nand2b_2 W5_sg13g2_nand3_1 W5_sg13g2_nand4_1 W5_sg13g2_nor2_1 W5_sg13g2_nor2_2 W5_sg13g2_nor2b_1 W5_sg13g2_nor2b_2 W5_sg13g2_nor3_1 W5_sg13g2_nor3_2 W5_sg13g2_nor4_1 W5_sg13g2_nor4_2 W5_sg13g2_o21ai_1 W5_sg13g2_or2_1 W5_sg13g2_or2_2 W5_sg13g2_or3_1 W5_sg13g2_tielo W5_sg13g2_xnor2_1 W5_sg13g2_xor2_1

$ cat broken_   # this is the filename after cellname rename
magic
tech ihp-sg13g2
timestamp 1751203273
<< end >>
```

Looks like the "(UNNAMED)" top cell is empty.  I was renaming just to get rid of the warning during `writeall modified` to check that was unreleated to the main issue.  The other top cell `tt_ihp_wrapper` has layout content.