RTimothyEdwards/magic #398: Crash when creating a MiM cap in sky130

smunaut, 2025-04-17:
I was trying to create a cap using the pcel generator 4x4 array of 9.5um x 9.5um cap and I get segfaults.

It's in `DBSrPaintArea` where the `GOTOPOINT` iterates too far, encountering an invalid value. Debug session below.

```
Thread 1 "wish" received signal SIGSEGV, Segmentation fault.
0x00007f9918e828b8 in DBSrPaintArea (hintTile=0x0, plane=0x55b6001106a0, rect=0x7ffd82604580, mask=0x7ffd82604590, 
    func=0x7f9918e9202f <areaCheck>, arg=0x7ffd826046a0) at DBtiles.c:378
378	    GOTOPOINT(tp, &start);
(gdb) print tp
$1 = (Tile *) 0xffffffffffffffff
(gdb) print plane->pl_hint
$2 = (Tile *) 0x7f9918674f38
(gdb) print plane->pl_hint->ti_lb
$3 = (struct tile *) 0x7f9918675168
(gdb) print plane->pl_hint->ti_lb->ti_lb
$4 = (struct tile *) 0x7f9918674918
(gdb) print plane->pl_hint->ti_lb->ti_lb->ti_lb
$5 = (struct tile *) 0x7f9918674678
(gdb) print plane->pl_hint->ti_lb->ti_lb->ti_lb->ti_lb
$6 = (struct tile *) 0x7f9918673b88
(gdb) print plane->pl_hint->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb
$7 = (struct tile *) 0x7f9918674058
(gdb) print plane->pl_hint->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb
$8 = (struct tile *) 0x7f99186732c8
(gdb) print plane->pl_hint->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb
$9 = (struct tile *) 0x7f9918673370
(gdb) print plane->pl_hint->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb->ti_lb
$10 = (struct tile *) 0xffffffffffffffff
(gdb) 
```

RTimothyEdwards, 2025-04-17:
@smunaut :

`GOTOPOINT` is supposed to move through the plane until it reaches a tile containing the point `start`.  There are two ways that this can go wrong:  (1) `start` is a point outside the bounds that magic defines as `MINFINITY` to `INFINITY`, or else (2) the database is corrupted.  It is always true that if you start at any tile in the plane and keep traveling by pointers in any direction, you will reach the tile that is defined as being placed at magic's definition of "infinity - 2", and its pointers (in the direction of "infinity") will be set to -1.

The values of interest passed to DBSrPaintArea() will be `*rect` (the total area to search), and `*mask`, which is the mask of types to look for.  Because the callback function is `areaCheck`, this search is coming from the DRC engine.  `rect` is an area of interest to search for a DRC error, and `mask` is the mask of tile types that, if found inside the area, constitute a DRC error.

If, from the point of the segfault, you go `up` two levels, you should be in `DRCbasic.c`, either line 978 or line 1319, and printing the contents of `*cptr` (the DRC rule structure), `tt` (the inside type), and `to` (the outside type), and `DRCCurStyle->DRCWhyList[cptr->drcc_tag]` (the printed text of the DRC rule).

smunaut, 2025-04-17:
Here you go, HTH

```
(gdb) bt
#0  0x00007fecaec828b8 in DBSrPaintArea (hintTile=0x0, plane=0x56053708a970, rect=0x7ffd97144890, mask=0x7ffd971448a0, func=0x7fecaec9202f <areaCheck>, arg=0x7ffd971449b0) at DBtiles.c:378
#1  0x00007fecaec94698 in drcTile (tile=0x7fecae5b5688, arg=0x7ffd971449b0) at DRCbasic.c:978
#2  0x00007fecaec82c9b in DBSrPaintArea (hintTile=0x0, plane=0x560536f25ce0, rect=0x7ffd97144a90, mask=0x7fecaf763e00 <DBAllTypeBits>, func=0x7fecaec92a07 <drcTile>, arg=0x7ffd971449b0) at DBtiles.c:441
#3  0x00007fecaec929b2 in DRCBasicCheck (celldef=0x560536d95d20, checkRect=0x7ffd97144a90, clipRect=0x7ffd97144ac0, function=0x7fecaec9b36d <drcPaintError>, cdata=0x56053420f910) at DRCbasic.c:405
#4  0x00007fecaec9e2bb in DRCInteractionCheck (def=0x560536d95d20, area=0x7ffd97144be0, erasebox=0x7ffd97144bf0, func=0x7fecaec9b36d <drcPaintError>, cdarg=0x56053420f910) at DRCsubcell.c:781
#5  0x00007fecaec9afe8 in drcCheckTile (tile=0x7fecae5b9e80, arg=0x0) at DRCcontin.c:713
#6  0x00007fecaec82c9b in DBSrPaintArea (hintTile=0x0, plane=0x560536f724c0, rect=0x7fecaeebae20 <TiPlaneRect>, mask=0x7fecaf763e40 <DBAllButSpaceBits>, func=0x7fecaec9ad92 <drcCheckTile>, arg=0x0) at DBtiles.c:441
#7  0x00007fecaec9ac3a in DRCContinuous () at DRCcontin.c:515
#8  0x00007fecb0c4d866 in TclServiceIdle () from /usr/lib64/libtcl8.6.so
#9  0x00007fecb0c29d08 in Tcl_DoOneEvent () from /usr/lib64/libtcl8.6.so
#10 0x00007fecb0d06bd2 in Tk_MainLoop () from /usr/lib64/libtk8.6.so
#11 0x00007fecb0d15b6a in Tk_MainEx () from /usr/lib64/libtk8.6.so
#12 0x0000560512583160 in main ()
(gdb) f 1
#1  0x00007fecaec94698 in drcTile (tile=0x7fecae5b5688, arg=0x7ffd971449b0) at DRCbasic.c:978
978			    DBSrPaintArea((Tile *) NULL,
(gdb) print *rect
$1 = {r_ll = {p_x = -8183, p_y = -1260}, r_ur = {p_x = 1260, p_y = 7120}}
(gdb) print *mask
No symbol "mask" in current context.
(gdb) print *cptr 
$2 = {drcc_dist = 268, drcc_mod = 0 '\000', drcc_cdist = 268, drcc_cmod = 0 '\000', drcc_mask = {tt_words = {1, 0, 0, 0, 0, 0, 0, 0}}, drcc_corner = {tt_words = {1, 0, 0, 90112, 0, 0, 0, 0}}, drcc_flags = 3, drcc_edgeplane = 12, 
  drcc_plane = 13, drcc_tag = 193, drcc_next = 0x0}
(gdb) print tt
$3 = 108
(gdb) print to
$4 = 0
(gdb) print DRCCurStyle->DRCWhyList[cptr->drcc_tag]
$5 = 0x56053748d8a0 "MiM cap spacing to unrelated metal3 < %d (capm.11)"
(gdb) 
```

smunaut, 2025-04-17:
Taken in frame #0 ( i.e. in DBSrPaintArea directly ): 

```
(gdb) print *rect
$9 = {r_ll = {p_x = -5931, p_y = 2488}, r_ur = {p_x = -5663, p_y = -1493881332}}
(gdb) print *mask
$10 = {tt_words = {4294967278, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295}}
```

That `rect` definitely looks suspicious.

dlmiles, 2025-04-17:
Is there a TCL script which when run does enough to run pcell generator to cause the same crash ?

Does disabling the DRC continuous prevent/delay the crash ?  if it prevents, does running on-demand DRC cause crash ? if not can DRC continuous be reenabled after the pcell generation has completed ?

smunaut, 2025-04-17:
Disabling DRC does prevent the crash, and I can re-enable it after the cell is created and it works fine.

Don't know enough TCL to create a script invoking PCells sorry. But I know the dimensions of the cap matter ... if I create a 2um x 2um cap it works fine, it I create a 9.5um x 9.5um one, it crashes.

RTimothyEdwards, 2025-04-17:
The area in DBSrPaintArea is extremely suspicious, as you noted, given that it represents a degenerate area (y_max < y_min).  That is something that would at least be easy to check for and ignore.  I would rather understand the circumstance in which it happens, because it's not supposed to happen.  The `rect.p_y` isn't even something obviously related to MINFINITY;  `-1493881332` is hex `0xa6f52e0c` which looks more like the Rect structure got overwritten with a pointer.

The fact that if you turn off DRC, create the device, then turn DRC back on is probably significant.  The DRC should not be running while the device is being generated;  maybe all that's needed is to make sure that DRC gets suspended at the beginning of device generation and resumed afterward (I would have thought I had done that, but if not, I should have).

dlmiles, 2025-04-22:
@smunaut please can you confirm more specific details of the sequence of actions to record in this issue so it can be looked at later before the specific details are forgotten.

Things like (but feel free to add relevant info) :
* magic version in use
* confirmation of the technology/PDK-version/pcell-generation-scripts version in use
* UI actions / TCL comands relating to the steps
* confirmation of the pcell generation process, maybe the input dialog has values that can be screenshot
* indication of the complexity of the project the cell is being generated for, or maybe confirmation with a large enough array/um area it will crash reliably as the first action into an empty project ?
Thanks

smunaut, 2025-04-24:
* Magic version : commit `6e83cbe2d3348971c6b4d95b6a28eb28d446a8d2`, tag: `8.3.525`
* PDK : Ciel revision `76760025e18728fb9705a4b2bf10dda0513b5bdd` of sky130A
* Video of the issue : https://github.com/user-attachments/assets/d5accb03-af4b-435e-b8e9-04640eb2bc5c
    * Don't pay attention to the hex numbers being printed, it's debug from trying to trace this issue ...
* It's an empty project, just started magic