RTimothyEdwards/magic #400: Difference between AppImage vs Source Built behaviour

SirJoeth3rd, 2025-04-23:
Hello. I've picked up a very small difference between the AppImage in the released folder and the version compiled from source. 

If I call magic --version using the compiled version it prints the version number and exits as expected according to the userguide. 

If I do the same using the appimage (./appimage-magic --version) it instead opens the magic console for further commands and doesn't print the version. 

This is potentially a non issue but i thought it worth raising, is this intentional perhaps? 

Kind regards.

dlmiles, 2025-04-24:
The `--version` processing maybe supplied by the Shell Script wrapper often installed as `/usr/local/bin/magic` not by the TCL environment or binary shell (wish) that has a TCL command called `version` that is different.

Maybe if you supply `-noconsole` argument such as `magic -noconsole -nowindow --version` it works more like you expect ?  But even that processing maybe done by the wrapper shell script.

There isn't a mode for executing a command and exiting `magic -noconsole -nowindow -x version`, only a file as a TCL script `magic -noconsole -nowindow script.tcl` which can contain:

```
version
exit
```

SirJoeth3rd, 2025-04-25:
Ah ok thank you for your reply. The magic command I used the first time was just a shell script wrapper, I just wanted to make sure I'm not accidentally using a different binary with unintentional behaviour. I'll close the issue, thanks again.

closed 2025-04-25 by SirJoeth3rd