RTimothyEdwards/magic #401: fail to compile magic on fedora 42

coryTuyaoqing, 2025-05-04:
Hi, i am trying to compile magic on fedora 42 and get this error:

              In file included from ../utils/tech.h:24,
                               from ../utils/utils.h:28,
                               from bpUtils.c:38:
              ../utils/magic.h:88:23: error: ‘bool’ cannot be defined via ‘typedef’
                 88 | typedef unsigned char bool;
                    |                       ^~~~
              ../utils/magic.h:88:23: note: ‘bool’ is a keyword with ‘-std=c23’ onwards
              ../utils/magic.h:88:1: warning: useless type name in empty declaration
                 88 | typedef unsigned char bool;
                    | ^~~~~~~
              In file included from ../utils/tech.h:24,
                               from ../utils/utils.h:28,
                               from bpEnum.c:38:
              ../utils/magic.h:88:23: error: ‘bool’ cannot be defined via ‘typedef’
                 88 | typedef unsigned char bool;
                    |                       ^~~~
              ../utils/magic.h:88:23: note: ‘bool’ is a keyword with ‘-std=c23’ onwards
              ../utils/magic.h:88:1: warning: useless type name in empty declaration
                 88 | typedef unsigned char bool;
                    | ^~~~~~~
              In file included from ../utils/tech.h:24,
                               from ../utils/utils.h:28,
                               from bpDump.c:37:
              ../utils/magic.h:88:23: error: ‘bool’ cannot be defined via ‘typedef’
                 88 | typedef unsigned char bool;
                    |                       ^~~~
              ../utils/magic.h:88:23: note: ‘bool’ is a keyword with ‘-std=c23’ onwards
              ../utils/magic.h:88:1: warning: useless type name in empty declaration
                 88 | typedef unsigned char bool;
                    | ^~~~~~~
              make[2]: *** [../rules.mak:25: bpUtils.o] Error 1
              make[2]: *** Waiting for unfinished jobs....
              make[2]: *** [../rules.mak:25: bpEnum.o] Error 1
              make[2]: *** [../rules.mak:25: bpDump.o] Error 1
              In file included from ../utils/tech.h:24,
                               from ../utils/utils.h:28,
                               from bpBins.c:39:
              ../utils/magic.h:88:23: error: ‘bool’ cannot be defined via ‘typedef’
                 88 | typedef unsigned char bool;
                    |                       ^~~~
              ../utils/magic.h:88:23: note: ‘bool’ is a keyword with ‘-std=c23’ onwards
              ../utils/magic.h:88:1: warning: useless type name in empty declaration
                 88 | typedef unsigned char bool;
                    | ^~~~~~~
              make[2]: *** [../rules.mak:25: bpBins.o] Error 1
              make[2]: Leaving directory '/home/cory/Downloads/magic-8.3.526/bplane'
              make[1]: *** [Makefile:52: modules] Error 1
              make[1]: Leaving directory '/home/cory/Downloads/magic-8.3.526'
              make: *** [Makefile:26: tcl] Error 2

and when I try to used c11 or c99 standard to compile it, i got this error:

              CmdSubrs.c: In function ‘CmdIllegalChars’:
              CmdSubrs.c:1231:14: error: implicit declaration of function ‘isascii’ [-Wimplicit-function-declaration]
               1231 |         if (!isascii(*p)) goto error;
                    |              ^~~~~~~
              make[2]: *** [../rules.mak:23: CmdSubrs.o] Error 1
              make[2]: *** Waiting for unfinished jobs....
              make[2]: Leaving directory '/home/cory/opensrc_eda/src/magic_src/commands'
              make[1]: *** [Makefile:52: modules] Error 1
              make[1]: Leaving directory '/home/cory/opensrc_eda/src/magic_src'
              make: *** [Makefile:26: tcl] Error 2

RTimothyEdwards, 2025-05-04:
That's a new one for me.

You should be able to work around the 1st problem by changing `utils/magic.h` line 88 to:

```
#ifndef bool
typedef unsigned char bool
#endif
```

Try that, and if it works for you I will update the source code.  Why your follow-up issue occurs, I have no idea.  `isascii()` should be defined in `ctype.h`, which is included at the top of `CmdSubrs.c`.  So I can't see why it would raise an error.

coryTuyaoqing, 2025-05-04:
Thanks very much for the advice.

I have try 

    #ifndef bool
    typedef unsigned char bool
    #endif
    
but it shows the same errors.

However, i eventually compile successfully by replacing gcc 15 with gcc 14. Fedora 42 by default install the newest version of gcc, which should be the reason why it doesn't work on magic.

RTimothyEdwards, 2025-05-04:
@dlmiles :  I upgraded to Fedora 41 a few months ago, so this is something that just popped up with Fedora moving to version 42.  Do you know what the proper fix should be?

dlmiles, 2025-05-04:
See https://github.com/RTimothyEdwards/magic/pull/362   (for the ground work on this one)

add into autoconf AC_HEADER_STDBOOL and regenerate (ensure to regenerate with autoconf-2.69, this is a good version to continue using with lots of Linux distro version compatibility, you may find F42 uses bleeding edge versions of things, bleeding-edge is not a good choice to use for autoconf, RHEL9 still ships/uses 2.69 by default)

```
#ifdef HAVE_STDBOOL_H
 #include <stdbool.h>
#else
 // the solution in the autoconf documentation around use of AC_HEADER_STDBOOL
 // is better and most compatible using _Bool type name etc...., but it replaced this line in magic.h
 typedef unsigned char bool;
#endif
```

autoconf docs and snippet here https://www.gnu.org/software/autoconf/manual/autoconf-2.69/html_node/Particular-Headers.html  Ctrl-F (find) AC_HEADER_STDBOOL

The header has been around since C99 and I don't think magic needs to target systems that need to support older compilers, those Unix systems should have access to a C99+ compatible GCC by now anyway.
So there is an argument to just replace the line with the #include header.

I this is still open in a few week I can make a patch for the above.

dlmiles, 2025-05-04:
> That's a new one for me.
> ...
> Try that, and if it works for you I will update the source code. Why your follow-up issue occurs, I have no idea. `isascii()` should be defined in `ctype.h`, which is included at the top of `CmdSubrs.c`. So I can't see why it would raise an error.

With `isascii()` API  removal, I have not got my notes to hand to cite related links, basically a C Standard committee decided to remove it.  So if you follow strict C23 you get error.