RTimothyEdwards/magic #409: nfet_05v0_nvt extraction issues.

DalinEM-lab, 2025-07-04:
Hi
I’m using the nfet_05v0_nvt device in my circuit, and while simulation in Xschem works correctly, I’m encountering an issue during layout extraction in Magic. Specifically, Magic is not extracting the transistor properly, it seems to treat the source and drain as a single net, likely due to the near-zero threshold voltage. (Please refer to the attached netlists on the left for reference.)
Has anyone encountered this issue before or know how to resolve it?

[image]
[image]

Thanks and Regards
Dalin

DalinEM-lab, 2025-07-06:
Hi, 
I tried using flatglob "sky130_fd_pr_*[A-Z]*", but it returns the error: "invalid command name 'A-Z'". I also tried flattening first and then extracting, but the resulting SPICE netlist only includes the source and drain terminals, the drain net  is missing. Please refer to the attached image for more details

[image]

d-m-bailey, 2025-07-07:
@DalinEM-lab Sorry about that. It comes from my poor memory of tcl syntax.
Try `flatglob {sky130_fd_pr__*[A-Z]*}`

DalinEM-lab, 2025-07-07:
@d-m-bailey flatglob {sky130_fd_pr__*[A-Z]*} is not working giving invalid command error. flatten glob {sky130_fd_pr__*[A-Z]*} is not giving any error. However, still magic is not extracting the drain net. 
 NGSPICE file created from test_nvtm.ext - technology: sky130A

.subckt sky130_fd_pr__nfet_05v0_nvt_2BXH7R a_n1000_n188# a_n1192_n322# a_1000_n100#
+ a_n1058_n100#
X0 a_1000_n100# a_n1000_n188# a_n1058_n100# a_n1192_n322# sky130_fd_pr__nfet_05v0_nvt ad=0.29 pd=2.58 as=0.29 ps=2.58 w=1 l=10
.ends

.subckt test_nvtm vb3 vss
Xsky130_fd_pr__nfet_05v0_nvt_2BXH7R_0 vss vss vb3 vb3 sky130_fd_pr__nfet_05v0_nvt_2BXH7R
.ends

d-m-bailey, 2025-07-07:
@DalinEM-lab extended apologizes.

`flatglob` is actually a directive used before reading gds. It doesn't work on data that is already in magic format.

If you wanted to try it, you could first export your design as gds, and then restart magic and run the following commands
```
gds flatglob {sky130_fd_pr__*[A-Z]*}
gds read <your_gds_file>
load <top_cell>
select <top_cell>
<extraction_commands>
```

DalinEM-lab, 2025-07-07:
@d-m-bailey   Thank you for your response. Unfortunately, the issue still persists. 

[image]

* NGSPICE file created from test_nvtm.ext - technology: sky130A

.subckt sky130_fd_pr__nfet_05v0_nvt_2BXH7R a_n1000_n188# a_n1192_n322# a_1000_n100#
+ a_n1058_n100#
X0 a_1000_n100# a_n1000_n188# a_n1058_n100# a_n1192_n322# sky130_fd_pr__nfet_05v0_nvt ad=0.29 pd=2.58 as=0.29 ps=2.58 w=1 l=10
.ends

.subckt test_nvtm vb3 vss
Xsky130_fd_pr__nfet_05v0_nvt_2BXH7R_0 vss vss vb3 vb3 sky130_fd_pr__nfet_05v0_nvt_2BXH7R
.ends

d-m-bailey, 2025-07-07:
@DalinEM-lab Great! Looks like it worked. Can you post the extracted netlist and the gds? 

The warning that's being displayed shows up on mos capacitors when the source and drain are connected to the same net.

DalinEM-lab, 2025-07-07:
Yes, the command worked. However, due to the zero threshold of the nfet_05v0_nvt, Magic is treating both the source and drain as a single net, even though, in my circuit source and drain are connected to different nets.  Which means that the extraction is incorrect, causing the LVS to fail and can't do post-layout simulations. Any idea how to resolve this issue?

d-m-bailey, 2025-07-08:
@DalinEM-lab np. We've all been there. Can you close this issue?

DalinEM-lab, 2025-07-08:
@d-m-bailey , Yes

closed 2025-07-08 by DalinEM-lab