RTimothyEdwards/magic #425: Install on MacOS Sequoia

danielw-etched, 2025-07-24:
I followed the README for MacOS using brew.  I made sure my path and DISPLAY are set as indicated.

`xterm` runs fine from the terminal.  A simple Tk script runs fine from the same terminal:

> import tkinter as tk
> from tkinter import ttk
> 
> def button_click():
>     """Function to be called when the button is clicked."""
>     label.config(text="Button clicked!")
> 
> root = tk.Tk()
> root.title("Simple Tk Demo")
> root.geometry("300x200")  # Set window size
> 
> label = ttk.Label(root, text="Hello, Tkinter!")
> label.pack(pady=20)  # Add padding
> 
> button = ttk.Button(root, text="Click Me", command=button_click)
> button.pack(pady=10)
> 
> root.mainloop()
 

But each time I try to start `magic` in the same terminal, it crashes in the same place in libX11:

> Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
> 0   libX11.6.dylib                	       0x112e12e40 XDefaultColormap + 12
> 1   tclmagic.dylib                	       0x114c11de8 GrTkInit + 296 (grTk1.c:411)
> 2   tclmagic.dylib                	       0x114c145b8 x11SetDisplay + 972 (grTk1.c:1281)
> 3   tclmagic.dylib                	       0x114c04dac GrSetDisplay + 548 (grMain.c:299)
> 4   tclmagic.dylib                	       0x114c8d7d8 mainInitAfterArgs + 516 (main.c:618)
> 5   tclmagic.dylib                	       0x114d7bba4 _magic_initialize + 644 (tclmagic.c:651)
> 6   libtcl8.6.dylib               	       0x1030eaff8 TclInvokeStringCommand + 136
> 7   libtcl8.6.dylib               	       0x1030ec778 TclNRRunCallbacks + 80
> 8   libtcl8.6.dylib               	       0x103177ef0 ChildEval + 132
> 9   libtcl8.6.dylib               	       0x103175a60 NRInterpCmd + 1704
> 10  libtcl8.6.dylib               	       0x1030ec778 TclNRRunCallbacks + 80
> 11  libtcl8.6.dylib               	       0x1030ed51c TclEvalEx + 1668
> 12  libtcl8.6.dylib               	       0x10318c4d0 Tcl_FSEvalFileEx + 636
> 13  libtk8.6.dylib                	       0x102f8a130 Tk_MainEx + 1248
> 14  wish8.6                       	       0x102d8ee2c main + 56
> 15  dyld                          	       0x18cd28274 start + 2840

dlmiles, 2025-07-24:
It appears from feedback the brew version of Tcl/Tk and X11 are not compatible with each other.
Also the version of X11 headers that are installed with the Tcl/Tk package is also incomplete, not complete enough for magic to use.

So the solution seems to be to install the X11 from brew.

Then build a local copy of Tcl/Tk on top of that X11 from brew.

Then build a local copy of magic using that combination of X11/Tcl/Tk.

Related background reading:
https://github.com/RTimothyEdwards/magic/issues/280

There maybe links to other recipes within that thread but some of the comment authors, such as macports.

Please respond back to this issue with any success or failure you found, Thanks.

I hope to look at having a DMG release image built via GitHub available for easier download and access, during the next look at GitHub Mac which will need maintenance in the next few months, due to EOL platforms in CI.
The DMG looks to see if it is possible to contain X11/Tcl/Tk/magic that are all compatible with each other.

danielw-etched, 2025-07-24:
That does not make complete sense to me, because I can run a Tk app just
fine, but I will try the X11 install.

Daniel

danielw-etched, 2025-07-25:
% brew install x11

==> Downloading https://formulae.brew.sh/api/formula.jws.json

==> Downloading https://formulae.brew.sh/api/cask.jws.json

Warning: No available formula with the name "x11".

==> Searching for similarly named formulae and casks...

==> Formulae

libx11 ✔

                                                              x11vnc

To install libx11 ✔, run:

  brew install libx11 ✔

% brew install libx11

Warning: libx11 1.8.12 is already installed and up-to-date.

To reinstall 1.8.12, run:

  brew reinstall libx11

% brew reinstall libx11

==> Downloading https://ghcr.io/v2/homebrew/core/libx11/manifests/1.8.12

Already downloaded:
/Users/danielw/Library/Caches/Homebrew/downloads/8510ddcf75e7c4313fe4f8b0999d667b22f2ae0b850bc80ba53becb609335934--libx11-1.8.12.bottle_manifest.json

==> Fetching libx11

==> Downloading
https://ghcr.io/v2/homebrew/core/libx11/blobs/sha256:57ccd55767fd6ffc7d32940d1c59d59b42c82abff4abf795b4ca0df489da42f1

Already downloaded:
/Users/danielw/Library/Caches/Homebrew/downloads/ac485b35b0a8bc039691451468442fe41812c2fb6850df986dcb9da73b5b3e71--libx11--1.8.12.arm64_sequoia.bottle.tar.gz

==> Reinstalling libx11

==> Pouring libx11--1.8.12.arm64_sequoia.bottle.tar.gz

🍺  /opt/homebrew/Cellar/libx11/1.8.12: 1,043 files, 7MB

==> Running `brew cleanup libx11`...

==> No outdated dependents to upgrade!

No change:  I can run `xterm` just fine.  I can run my Tk button program
just fine.  `magic` crashes in the same place.  I even tried running
`xterm` and starting `magic` from the `xterm`.  `magic` crashed in the same
place.

Daniel

dlmiles, 2025-07-25:
Which part needs clarification ?

X11 the system comes in a server part and a client library part.  There is a network protocol between the two parts and some amount of interchangeability / interop across that network layer.

The libX11 client library part exists both inside the `tcl-tk` brew package and the `xquartz`.

The TclTk (from brew) does come with a version of X11 client libraries that is sufficient to run all the features of the Tcl/Tk system , so long as you stay inside the Tcl script system and widgets, but the Tcl/Tk system also allows C API customization to built on.

But Magic requires Tcl and Tk and X11 (full featured, not a cut down version that only works with a specific version of Tcl/Tk packages up inside brew).
Magic makes use of X11 APIs directly, as well as via Tk C APIs and Tcl scripts, so needs a full and complete suite of libX11 APIs and symbols as per the X11R6 specification.
Magic has support for a number of X extensions relating to graphics drivers and other matters that is uses directly (not via Tk).
Magic also has some customized Tk widgets.

I am claiming it appears the version of X11 client libraries bundled with Tcl/Tk only support running Tcl/Tk scripts.
I am claiming when you install a full and complete X11 implementation (such as `xquartz`) which includes a full set of APIs, you can not mix and match the DSOs loaded at runtime as only one set of libX11 client DSOs can be loaded into the process that magic/Tcl/Tk all share.
But the libX11 from xquartz is the one magic needs (as it is full featured),
But the libX11 from tcl-tk is the one tcl-tk needs (as it was built and bundled with it)
But they are not compatible at some level, after complication, so it is not possible to substitute, and this causes the crash on startup.

This is my best understanding (which maybe incorrect) based on the information provided.
Unfortunately I do not have access to a MacOS system, only the GitHub CI, which can be used to confirm the project builds as can be seen https://github.com/RTimothyEdwards/magic/actions/workflows/main-macos.yml but it has not been useful to validate the resulting binary actually runs (without a crash on startup).

So the resolution is to build your own copy of Tcl/Tk against a full and complete X11 implementation (such as `xquartz`).
Then build magic against that version of Tcl/Tk.  Now the resulting magic/Tcl/Tk all use the same X11 client library DSO.
I believe a comment or link in the other thread might show how that was done from someone who reported success.

You own stack trace of the crash, indicates it crashes early on startup `_magic_initialize` on what maybe the very first direct X11 API call to `XDefaultColormap` made by magic code `grTk1.c:411`.  Which is congruent with my commentary and understanding above.

Please research the link #280 with feedback from other MacOS users.
HTH