RTimothyEdwards/magic #451: Wrong colors in the 3D visualization of a PMOS transistor

OtacilioNeto, 2025-09-26:
When doing the 3D visualization of a PMOS transistor, Magic is displaying the colors in the visualization in the same way as the NMOS transistor. Test conducted with the Sky130 PDK.

[image]

RTimothyEdwards, 2025-09-26:
That's because the 3D view is showing you actual GDS mask layers, not magic's layers (implants like NSDM and PSDM are not shown) so it is showing only "DIFF".

You can toggle between the magic layer view and the mask layer view with the "C" key (see http://opencircuitdesign.com/magic/commandref/wind3d/cif.html).

RTimothyEdwards, 2025-09-26:
Note, however, that the magic layer view will size contacts by the contact area, not the contact cut, and transistors will appear as a solid block between diffusion and poly without any space under the gate, which is why the GDS layers appear by default.

OtacilioNeto, 2025-09-26:
Ok. Thank you very much for your assistance.